adokter/vol2bird #43: ubunta install instructions fail

adokter, 2016-06-02:
@jspaaks @LourensVeen: Maarten Reyniers volgde de INSTALL.md, maar liep vast:

email 1 jun 2016:
Hier dus een meer technische mail in verband met de installatie van vol2bird.
Ik ben vol2bird momenteel aan het installeren op een Ubuntu machine (64 bit) met Ubuntu 16.04.
De machine is een virtuele machine die ik volledig nieuw heb aangemaakt met VirtualBox (de host is een Mac).
Ik volg je installatie instructies van https://github.com/adokter/vol2bird/blob/master/INSTALL.md
Het loopt een eerste maal mis bij de installatie van Pillow (pip install Pillow).
De package dat eerst moet geïnstalleerd worden is python-dev (dit zit niet standaard in de distro).
Ergens zou je dus moeten toevoegen
sudo apt-get install python-dev
Dan gaat de installatie van Pillow wel door.

Daarna is er een probleem met de compilatie van rave.
Ik ben geen python specialist, maar blijkbaar is er een conflict met numpy.
Zie hieronder waar de make van rave fout loopt.
De fout heeft iets te maken met de virtual environment, want bij een system-wide installatie van de python packages lukt het me wel om rave te compileren (en aansluitend ook vol2bird). Maar ik zou graag jouw instructies volgen want met de virtual environment is het net iets eleganter.

Heb jij een idee wat ik moet doen om rave verder te kunnen compileren onder de virtual environment?

Vriendelijke groeten,
Maarten

```
...
make -C pyapi
make[2]: Entering directory '/home/maarten/birdsoft/src/rave/librave/pyapi'
In file included from /home/maarten/birdsoft/.venv/local/lib/python2.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:4:0,
from /home/maarten/birdsoft/.venv/local/lib/python2.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:18,
from /home/maarten/birdsoft/.venv/local/lib/python2.7/site-packages/numpy/core/include/numpy/arrayobject.h:4,
from getpy.h:11,
from getpy.c:7:
/home/maarten/birdsoft/.venv/local/lib/python2.7/site-packages/numpy/core/include/numpy/npy_common.h:319:2: error: #error Must use Python with unicode enabled.
#error Must use Python with unicode enabled.
^
cp: cannot stat '.dep/getpy.d': No such file or directory
/bin/sh: 3: cannot open .dep/getpy.d: No such file
x86_64-linux-gnu-gcc -c -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -pthread -fPIC -DCREATE_ITRUNC -I. -I../toolbox -I/home/maarten/birdsoft/.venv/local/lib/python2.7/site-packages/numpy/core/include/numpy -I/home/maarten/birdsoft/.venv/include/python2.7 -I/home/maarten/birdsoft/opt/hlhdf/include -I/usr/include/hdf5/serial/ getpy.c
In file included from getpy.c:7:0:
getpy.h:10:20: fatal error: Python.h: No such file or directory
compilation terminated.
Makefile:27: recipe for target 'getpy.o' failed
make[2]: *** [getpy.o] Error 1
make[2]: Leaving directory '/home/maarten/birdsoft/src/rave/librave/pyapi'
Makefile:24: recipe for target 'libpyapi' failed
make[1]: *** [libpyapi] Error 2
make[1]: Leaving directory '/home/maarten/birdsoft/src/rave/librave'
Makefile:29: recipe for target 'librave' failed
make: *** [librave] Error 2
```

LourensVeen, 2016-06-09:
We suspect that when you do `pip install numpy`, numpy gets compiled, and if you don't have libicu installed, then it will get installed without Unicode support, which then later results in the above error. To fix this, do `sudo apt-get install libicu55 libicu-dev` (on Ubuntu 16.04, it's `libicu52` on 14.04) before `pip install numpy`.

Please let us know if that fixes it for you.

adokter, 2016-06-16:
@LourensVeen yep this fixes it. I updated the install instructions. Closing the issue

closed 2016-06-16 by adokter