adokter/vol2bird #58: pyvol2bird has an unused argument

adokter, 2016-07-11:
When [checking for arguments](https://github.com/adokter/vol2bird/blob/344fbd2e693b1a10a6e3105b4d1619d64193e8fe/pyvol2bird/pyvol2bird.c#L157) both the vertical profile object and an unused double called `dBlim` has to be passed. We can get rid of this argument

adokter, 2016-07-11:
After adjusting, also the call in the [pgf](https://github.com/adokter/vol2bird/blob/cc8bd4c919d342e8df3ad2cd24b0336af9316121/pgfplugin/rave_pgf_vol2bird_plugin.py#L79) needs to updated

LourensVeen, 2016-07-12:
Don't forget to update the test case too, and run it to make sure everything still works (or at least as well as it did, there is still the 'dens' issue).

adokter, 2016-07-14:
closed by committ https://github.com/adokter/vol2bird/commit/cf32af75f74a7dc4f96c444ec6084481d260e642

closed 2016-07-14 by adokter