adokter/vol2bird #59: extract radar unique identifier for parsing config files

adokter, 2016-07-12:
Easiest to use this python function:
```
>>> polarvolume = _raveio.open("seang_vp_20160712093000.h5").object
>>> NOD=odim_source.NODfromSource(polarvolume)
>>> NOD
>>> 'seang'
```

LourensVeen, 2016-07-12:
I think it's better to do this in C, so that vol2bird does the same thing as the PGF plug-in.

There's a problem in that you don't have the radar name during setup, while that is where the config file is read. That can probably be solved by reading the default options during setup, then reading in another file with radar-specific settings at the start of the vol2bird() function, on top of the defaults. A better option may be to use radar-specific sections in the config file (Jurriaan seemed to remember that he'd chosen libconfuse because it had that capability), load the whole thing, and then just use the right ones during processing.

That would also let you do setup once and then call vol2bird() multiple times with files from different radars, at least if we fix the memory management issues.

adokter, 2016-07-15:
@LourensVeen agree, but the problem is that the radar identifier can be coded in many ways. The suggested python tool uses many tests and a  look-up table (in xml) to find the correct identifier - will take a lot of time to code this ourselves.

adokter, 2016-07-15:
To do:
* Add optional configuration file option to command line tool, as in https://www.gnu.org/software/libc/manual/html_node/Getopt-Long-Option-Example.html#Getopt-Long-Option-Example
* add argument to [vol2birdLoadConfig](https://github.com/adokter/vol2bird/blob/8d99fedab24e30aca00d97bde686b76c912774cc/lib/libvol2bird.c#L3763) that parses the filename of the configuration file and its call of [ReadUserConfigOptions](https://github.com/adokter/vol2bird/blob/344fbd2e693b1a10a6e3105b4d1619d64193e8fe/lib/libvol2bird.c#L2117)
* add the same argument to [readUserConfigOptions](https://github.com/adokter/vol2bird/blob/344fbd2e693b1a10a6e3105b4d1619d64193e8fe/lib/libvol2bird.c#L2117) and its [call of cfg_parse]
(https://github.com/adokter/vol2bird/blob/344fbd2e693b1a10a6e3105b4d1619d64193e8fe/lib/libvol2bird.c#L2155)
* if `cfg_parse` cannot read the file, make it default to the standard `options.conf`
* add the same argument to [PyVol2Bird_New](https://github.com/adokter/vol2bird/blob/344fbd2e693b1a10a6e3105b4d1619d64193e8fe/pyvol2bird/pyvol2bird.c#L77) C function, and this [call](https://github.com/adokter/vol2bird/blob/344fbd2e693b1a10a6e3105b4d1619d64193e8fe/pyvol2bird/pyvol2bird.c#L88) in its body.
* add a `odim_source.NODfromSource(polarvolume)` call to [_pyvol2bird_new](https://github.com/adokter/vol2bird/blob/344fbd2e693b1a10a6e3105b4d1619d64193e8fe/pyvol2bird/pyvol2bird.c#L130-L133) python function

adokter, 2016-07-15:
Make `vol2bird` load configuration following this hierarchy config files:
1. default
2. country
3. radar