adokter/vol2bird #60: fetch data from SMHI

adokter, 2016-07-13:


adokter, 2016-07-13:
Using a crontab on horizon, running the following script
```
adokter1@horizon:~/bin$ cat fetch_vp_smhi.sh 
#!/bin/bash
yday=`date -d "yesterday 13:00 " '+%Y%m%d'`
today=`date '+%Y%m%d'`
dir=`mktemp -d` && cd $dir
wget --password=<mypassword> --quiet volbird@co-0181.c.ipeer.se:/data/*.h5

cadaver -t <<EOF
open https://beehub.nl/ENRAM
mkdir $yday
cd $yday
mput *$yday*.h5
cd ..
mkdir $today
cd $today
mput *$today*.h5
quit
EOF

rm -rf $dir
```

adokter, 2016-07-13:
To enable the script, both `cadaver` and `openssl` needed to be installed.

* To add ssl support, first install openssl
* then install [cadaver](http://www.webdav.org/cadaver/), which in my case required several flags because I am installing not as a super user:
 ./configure --prefix=/home/adokter1/ --with-ssl=openssl --with-libs=/home/adokter1:/usr/lib --with-ca-bundle=/etc/ssl/certs/ca-certificates.crt

closed 2016-07-13 by adokter

LourensVeen, 2016-07-13:
For anyone wishing to set up a similar system: note that it is a bad idea to install OpenSSL locally, as it receives regular security updates, and you're unlikely to keep up with those yourself, resulting in security and/or compatibility issues. I strongly recommend having your system administrator install OpenSSL and Cadaver from packages, so that they'll be kept up-to-date with the rest of the system.

adokter, 2016-07-13:
@LourensVeen can you do that? - still having admin rights on horizon?

adokter, 2016-07-13:
Fixed it, now using cadaver from packages