adokter/vol2bird #72: Makefile doesn't detect changes to files

adokter, 2016-08-11:
The [Makefile](https://github.com/adokter/vol2bird/blob/master/lib/Makefile) of libvol2bird doesn't pick up changes to source files. When I updated a source file and run `make`, it always gives `make: Nothing to be done for 'all'`.

I need to do `make clean` followed by `make` to compile the sources, which isn't how Make should behave

@LourensVeen: any ideas what's wrong in the Makefile code?

LourensVeen, 2016-08-18:
Looks like either there was no dependency generation, or it got removed in the recent refactor. I've added dependencies in the Makefiles directly now, which is slightly less nice than generating them automatically, but given the small number of them and the rather cryptic syntax you get with automatic dependency generation this is probably the better solution for now.

adokter, 2016-08-21:
Closed by commit https://github.com/adokter/vol2bird/commit/e930594466cfc97c762336c38f5c99899920d5b7

closed 2016-08-21 by adokter