adokter/vol2bird #95: Why is HGHT data5 instead of data1?

peterdesmet, 2017-06-16:
According to the [ODIM bird profile specification](https://github.com/adokter/vol2bird/wiki/ODIM-bird-profile-format-specification), the HGHTs can be found in `dataset1/data1`:

```
/dataset1/data1/what
/dataset1/data1/what/quantity "HGHT"
/dataset1/data1/data                 1x20 array #Dataset containing profile heights
```

However, for the h5 files I have (e.g. [this one](https://github.com/enram/vp-processing/blob/master/data/raw/example/seang_vp_20161003T2000Z.h5)) I noticed the HGHT is in `dataset1/data5` and `quantity` is not included:

```r
> h5read("../data/raw/example/seang_vp_20161003T2000Z.h5", "/dataset1")

List of 16

...
$ data5 :List of 2
  ..$ data: num [1, 1:25] 0 200 400 600 800 1000 1200 1400 1600 1800 ...
  ..$ what: NULL
...
```

@adokter

- Why is this in `data5`?
- Is this consistent across h5 files?
- How does bioRad know which variable is in which data group? Is this hard coded?

peterdesmet, 2017-06-16:
It's even the case for the example file in bioRad: https://github.com/adokter/bioRad/blob/master/inst/extdata/profile.h5

adokter, 2017-06-16:
@peterdesmet:
* The reason for `HGHT` to be in `data5` is fully arbitrary, and set by the [Rave](http://git.baltrad.eu/manual/rave/) C-library used by vol2bird.
* The ordering of the quantities in a vertical profile cannot be controlled, and (I think) only reflects the order in which the quantities were generated by the Rave library. You cannot assume it to be consistent across h5 files
* Therefore it is essential that you don't use the ordering for extracting specific quantities from the vertical profile. Instead, use the attribute `quantity` of the `dataset1/dataX/what` group, which contains the name of the quantity as a string.

closed 2017-06-16 by adokter