adokter/vol2bird #108: poor dealiasing of certain German radial velocity data

adokter, 2018-11-02:
Dealiasing fails for RAD_NEU_10557_VOL_NA_201810312115.h5, with radial velocity fit returning many NA. Nyquist velocity of 8 m/s seems correct. Need to dig deeper to find reason for poor performance.

```
Warning: no user configuration file 'options.conf' found. Using default settings ...
Warning: using radar wavelength stored for scan 1 (5.328045 cm) for all scans ...
Warning: no dual-pol moments not found, entering SINGLE POL mode
Warning: Nyquist interval attribute not found for scan 1, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 2, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 3, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 4, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 5, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 6, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 7, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 8, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 9, using radial velocity offset (8.0 m/s) instead 
Warning: Nyquist interval attribute not found for scan 10, using radial velocity offset (8.0 m/s) instead 
Warning: radial velocities will be dealiased...
# vol2bird Vertical Profile of Birds (VPB)
# source: WMO:10557
# polar volume input: RAD_NEU_10557_VOL_NA_201810312115_2.h5
# date   time HGHT    u      v       w     ff    dd  sd_vvp gap dbz     eta   dens   DBZH   n   n_dbz n_all n_dbz_all
20181031 2115    0    nan    nan     nan   nan   nan    nan T    nan    nan    nan    nan     0     0     0     0
20181031 2115  200    nan    nan     nan   nan   nan    nan T    nan    nan    nan    nan     0     0     0     0
20181031 2115  400    nan    nan     nan   nan   nan    nan T    nan    nan    nan    nan     0     0     0     0
20181031 2115  600    nan    nan     nan   nan   nan    nan T    nan    nan    nan    nan     0     0     0     0
20181031 2115  800    nan    nan     nan   nan   nan   2.94 T  -8.12   54.5   0.00   3.81    14   106  1130  3102
20181031 2115 1000    nan    nan     nan   nan   nan   2.58 T -10.30   33.0   0.00   0.99   104  1327  2167  6874
20181031 2115 1200    nan    nan     nan   nan   nan   2.67 T  -7.32   65.5   0.00   2.06    38   237  2698  6448
20181031 2115 1400    nan    nan     nan   nan   nan   2.46 T  -6.86   72.8   0.00   2.00    41   194  3332  6063
20181031 2115 1600    nan    nan     nan   nan   nan   2.78 T  -5.33  103.4   0.00   1.63    69   284  2667  4667
20181031 2115 1800    nan    nan     nan   nan   nan   2.60 T  -3.42  160.4   0.00   1.75    46   138  2658  3949
20181031 2115 2000 -11.62  -8.66   -3.02 14.49 233.3   2.39 F  -2.73  188.4  17.12   0.00   115   467  2570  3745
20181031 2115 2200  -9.90  -6.53   -3.67 11.85 236.6   2.28 F  -5.19  106.8   9.71  -2.27   251   779  2054  3493
20181031 2115 2400 -10.34  -4.68   -4.71 11.35 245.6   2.16 F  -5.78   93.3   8.49  -4.34   497  1618  1182  2776
20181031 2115 2600 -10.34  -4.76    1.65 11.38 245.3   1.99 F -10.91   28.6   0.00  -8.56   291  2177   558  2771
20181031 2115 2800    nan    nan     nan   nan   nan    nan T -17.89    5.7   0.00 -15.20    57  1631   103  1747
20181031 2115 3000    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  2123     0  2124
20181031 2115 3200    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1747     0  1747
20181031 2115 3400    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
20181031 2115 3600    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1407     0  1407
20181031 2115 3800    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
20181031 2115 4000    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1440     0  1440
20181031 2115 4200    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
20181031 2115 4400    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
20181031 2115 4600    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0   360     0   360
20181031 2115 4800    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
```

adokter, 2018-11-02:
Issue is related to the fact that many gates have no valid reflectivity value. Default compilation of vol2bird requires data in all 8 azimuthal sectors, as set by `NBINSGAP` in constants.h

Setting `NBINGSGAP=1`, which disables checking for azimuthal gaps, gives better results.

```
# vol2bird Vertical Profile of Birds (VPB)
# source: WMO:10557
# polar volume input: RAD_NEU_10557_VOL_NA_201810312115_2.h5
# date   time HGHT    u      v       w     ff    dd  sd_vvp gap dbz     eta   dens   DBZH   n   n_dbz n_all n_dbz_all
20181031 2115    0    nan    nan     nan   nan   nan    nan T    nan    nan    nan    nan     0     0     0     0
20181031 2115  200    nan    nan     nan   nan   nan    nan T    nan    nan    nan    nan     0     0     0     0
20181031 2115  400    nan    nan     nan   nan   nan    nan T    nan    nan    nan    nan     0     0     0     0
20181031 2115  600    nan    nan     nan   nan   nan    nan T    nan    nan    nan    nan     0     0     0     0
20181031 2115  800 -18.64 -37.83 -3851.03 42.17 206.2   2.94 F  -8.12   54.5   4.95   3.81    14   106  1130  3102
20181031 2115 1000 -10.08  -6.10   -5.14 11.78 238.8   2.58 F -10.30   33.0   3.00   0.99   104  1327  2167  6874
20181031 2115 1200 -15.44  -5.95  -26.52 16.54 248.9   2.67 F  -7.32   65.5   5.95   2.06    38   237  2698  6448
20181031 2115 1400 -15.26  -4.57  -43.09 15.93 253.3   2.46 F  -6.86   72.8   6.62   2.00    41   194  3332  6063
20181031 2115 1600 -13.19  -7.78    5.07 15.32 239.5   2.78 F  -5.33  103.4   9.40   1.63    69   284  2667  4667
20181031 2115 1800 -16.46  -6.86  -28.21 17.84 247.4   2.60 F  -3.42  160.4  14.58   1.75    46   138  2658  3949
20181031 2115 2000 -11.62  -8.66   -3.02 14.49 233.3   2.39 F  -2.73  188.4  17.12   0.00   115   467  2570  3745
20181031 2115 2200  -9.90  -6.53   -3.67 11.85 236.6   2.28 F  -5.19  106.8   9.71  -2.27   251   779  2054  3493
20181031 2115 2400 -10.34  -4.68   -4.71 11.35 245.6   2.16 F  -5.78   93.3   8.49  -4.34   497  1618  1182  2776
20181031 2115 2600 -10.34  -4.76    1.65 11.38 245.3   1.99 F -10.91   28.6   0.00  -8.56   291  2177   558  2771
20181031 2115 2800 -10.44  -2.94   -0.32 10.85 254.3   1.78 F -17.89    5.7   0.00 -15.20    57  1631   103  1747
20181031 2115 3000    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  2123     0  2124
20181031 2115 3200    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1747     0  1747
20181031 2115 3400    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
20181031 2115 3600    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1407     0  1407
20181031 2115 3800    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
20181031 2115 4000    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1440     0  1440
20181031 2115 4200    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
20181031 2115 4400    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
20181031 2115 4600    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0   360     0   360
20181031 2115 4800    nan    nan     nan   nan   nan    nan T   -inf    0.0   0.00   -inf     0  1080     0  1080
```
Also, this file has many more valid VRADH data than it has DBZH data, so quality control on DBZH seems much stricter.

Have to decide whether we can make a change to accept VRADH data even when there is no valid DBZH data. That would be the preferred behaviour for this radar, but might get us in trouble with other stations.

adokter, 2018-11-02:
Two examples of radial velocity and reflectivity data for the same scan. Lots of DBZH gates have no valid data value

[image]
[image]

leijnse, 2018-11-12:
Strong filtering for DBZH by DWD seems to be the cause of the loss of data in DBZH. We're trying what would happen if we replace DBZH with TH (i.e., uncorrected data). This may work for radars that are not close to the sea (where we would expect significant clutter with non-zero Doppler velocity).

I need to check with DWD (or possibly our own internal converters) about the unexpected Nyquist interval values. With dual PRF (600/800 Hz), we would expect a much better range than 8 m/s.