adokter/vol2bird #120: Define vpts output format in tabular format

peterdesmet, 2019-04-10:
Which columns to include in an easy to use vpts text-based output?

adokter, 2019-04-16:
There is already a `as.data.frame` method for `vpts` objects in bioRad, that would be a good start:

```
> str(as.data.frame(example_vpts))
'data.frame':	48350 obs. of  23 variables:
 $ radar    : chr  "KBGM" "KBGM" "KBGM" "KBGM" ...
 $ datetime : POSIXct, format: "2016-09-01 00:02:00" "2016-09-01 00:02:00" "2016-09-01 00:02:00" ...
 $ height   : int  0 200 400 600 800 1000 1200 1400 1600 1800 ...
 $ u        : num  NaN NaN NaN 4.14 5.06 4.74 4.41 4.78 3.75 4.14 ...
 $ v        : num  NaN NaN NaN 3.84 0.24 -2.69 -4.09 -3.55 -3.03 -2.03 ...
 $ w        : num  NaN NaN NaN 12.2 15.2 ...
 $ ff       : num  NaN NaN NaN 5.65 5.07 5.45 6.02 5.96 4.82 4.61 ...
 $ dd       : num  NaN NaN NaN 47.2 87.2 ...
 $ sd_vvp   : num  NaN NaN 2.81 2.8 2.42 2.38 2.29 2.34 2.62 3.06 ...
 $ gap      : logi  TRUE TRUE TRUE FALSE FALSE FALSE ...
 $ dbz      : num  NaN NaN 1.54 3.36 -7.89 ...
 $ eta      : num  NaN NaN 30.8 46.9 3.5 0.5 0.2 0.1 0.1 0.1 ...
 $ dens     : num  NaN NaN 2.8 4.264 0.318 ...
 $ DBZH     : num  NaN NaN 3.77 0.5 -10.33 ...
 $ n        : int  0 0 326 9006 2313 652 990 407 247 154 ...
 $ n_dbz    : int  0 0 356 13442 11145 8851 8026 9945 11786 10878 ...
 $ n_all    : int  0 0 22485 65947 21321 11789 8434 2718 442 163 ...
 $ n_dbz_all: int  0 0 28416 104455 63542 38006 25583 20376 14716 11482 ...
 $ lat      : num  42.2 42.2 42.2 42.2 42.2 42.2 42.2 42.2 42.2 42.2 ...
 $ lon      : num  -76 -76 -76 -76 -76 ...
 $ day      : logi  FALSE FALSE FALSE FALSE FALSE FALSE ...
 $ sunrise  : POSIXct, format: "2016-09-01 10:32:54" "2016-09-01 10:32:54" "2016-09-01 10:32:54" ...
 $ sunset   : POSIXct, format: "2016-09-01 23:33:47" "2016-09-01 23:33:47" "2016-09-01 23:33:47" ...
```

* The day, sunrise and sunset fields are calculate by the `as.data.frame` method using `check_night()`, `sunrise()` and `sunset()` functions
* lat, lon fields are essentially replicates of the vpts attributes `example_vp$attributes$where$lat` and `example_vp$attributes$where$lon`, so could be removed too.

adokter, 2019-04-16:
Besides the actual profile data, we also need to define which attribute quantities are needed, which are typically the same for a set of `vp` forming a `vpts`

minimally required attributes that are currently accessed by various bioRad functions:
* `example_vp$attributes$how$beamwidth`
* `example_vp$attributes$how$rcs_bird`
* `example_vp$attributes$how$sd_vvp_thresh`
* `example_vp$attributes$how$rcs_bird`
* `example_vp$attributes$how$task_version`
* `example_vp$attributes$how$task`
* `example_vp$attributes$how$task_args` (this is a very long string containing all the settings of the vol2bird algorithm)
* `example_vp$attributes$how$wavelength`
* `example_vp$attributes$what$date`
* `example_vp$attributes$what$time`
* `example_vp$attributes$what$source` (from this the `example_vpts$radar` field is derived
* `example_vp$attributes$where$height` (this is the antenna height)
* `example_vp$attributes$where$interval` (altitude layer width)
* `example_vp$attributes$where$lat`
* `example_vp$attributes$where$lon`

peterdesmet, 2019-04-16:
Would the attributes be the same for a month of vpts data?

Ways to add attributes:

- repeating it for every row = increased file size
- adding it once above the header row in a csv = somewhat clunky
- having a separate metadata file = not directly associated
- expressing data as json, with attributes and data as different objects = probably best solution? Are there json libraries in C for vol2bird?

peterdesmet, 2019-04-16:
Oh, regarding option 3 (separate metadata file), we could use the [data package spec](https://frictionlessdata.io/data-packages/). The advantage over json is that it is well defined. I’ve already expressed vpts data as a data package for the nl/be case study:

peterdesmet, 2019-04-16:
And for option 2, http://csvy.org/ looks like an elegant solution in which metadata is added as yaml frontmatter. I quite like that, as super simple and human readible. Some libraries already support it.

peterdesmet, 2019-04-25:
Question: do we he have to report `dens` and `rcs_bird`? Could be calculated easily by user, since it is just: `eta / rcs_bird = dens` with `rcs_bird` a constant value.

adokter, 2019-04-25:
I have been adding `dens` to make things easy for users, as there are two steps involved that involve some background knowledge that especially new people likely won't have:

1) calculate `dens = eta / rcs_bird`
2) set `dens = 0` when `sd_vvp < sd_vvp_threshold`
Both `rcs_bird` and `sd_vvp_threshold` could be changed by the user, but we have been promoting as good practice to keep `rcs_bird`=11 cm^2 and `sd_vvp_threshold`=2 m/s.

My reasoning has been that these two steps are not that obvious, such that adding `dens` as a column is worth it.

bart1, 2019-10-23:
@BerendWijers you might want to follow this issue as well

adokter, 2020-11-06:
Attributes set by vol2bird:

- `how/beamwidth`
- `how/wavelength`
- `how/rcs_bird`
- `how/sd_vvp_thresh`
- `how/dealiased`
- `how/task`
- `how/task_version`
- `how/task_args`
- `how/comment`
- `how/minrange`
- `how/maxrange`
- `how/minazim`
- `how/maxazim`
- `how/clutterMap`
- `how/filename_pvol`
- `how/filename_vp`
- `how/vcp`: volume coverage pattern (NEXRAD)

adokter, 2020-11-06:
https://specs.frictionlessdata.io/