adokter/vol2bird #131: create R package

adokter, 2019-10-07:
adokter/bioRad#156 requires compilation of vol2bird within R, instead of accessing it through Docker.

To keep code bases separate, I suggest making an R-package for vol2bird that will be a dependency to bioRad (instead of integrating vol2bird C-code itself into bioRad)

R documentation for how to include external code:
https://cran.r-project.org/doc/manuals/r-release/R-exts.html#Non_002dR-scripts-in-packages

Hadley Wickham's tutorial on compiled code:
http://r-pkgs.had.co.nz/src.html

Examples of R packages based on C/C++ code:
* [xgboost](https://github.com/dmlc/xgboost/tree/master/R-package)
* [Rhdf5lib](https://github.com/grimbough/Rhdf5lib)
* [tibble](https://github.com/tidyverse/tibble)

Required steps:
- [ ] define git submodules for all library dependencies used by vol2bird. Note that we will need to install a package with submodules using [install_git](https://www.rdocumentation.org/packages/devtools/versions/1.13.6/topics/install_git) with `args="--recursive"` instead of [install_github](https://www.rdocumentation.org/packages/devtools/versions/1.13.6/topics/install_github).
- [ ] define configure and cleanup scripts, see https://cran.r-project.org/doc/manuals/r-release/R-exts.html#Configure-and-cleanup
- [ ] rewrite vol2bird's Makefile into an R-compatible Makevars file
- [ ] (need to check) register library functions we want to use in R, see [CRAN manual](https://cran.r-project.org/doc/manuals/r-release/R-exts.html#Linking-to-native-routines-in-other-packages) and [xgboost](https://github.com/dmlc/xgboost/blob/master/R-package/src/init.c) for examples. This seems also the process by which packages can share compiled code, e.g. see how [rhdf5](https://github.com/grimbough/rhdf5) uses [Rhdf5lib](https://github.com/grimbough/Rhdf5lib)

adokter, 2019-10-07:
See [this discussion](https://github.blog/2016-02-01-working-with-submodules/) of submodules. There is also a lot of [criticism](https://medium.com/@uttamkini/sharing-code-and-why-git-submodules-is-a-bad-idea-1efd24eb645d) of submodules out there.

adokter, 2019-10-08:
For Windows compilation, need to find out if we can cross-compile using [Rtools](https://cran.r-project.org/bin/windows/Rtools/)

bart1, 2020-04-15:
Quick update I have been able to make a small toy package that converts files, it though still depends on the libvol2bird.so file, lets see if i can include the source

bart1, 2020-04-15:
I just got a version working where the library is compiled directly into the r package. It might still be a challenge to get it cran compatible but it works well. I will upload it later today

bart1, 2020-04-15:
Here is a quick mockup: https://github.com/bart1/Rvol2bird  it is only tested on linux, and mostlikely does not directly compile on other OS's. Currently copies of the required libraries are included (no git subtrees). So it does not follow changes in the libaries

adokter, 2020-04-16:
Asked Simon Urbanek about proj4 header files in https://github.com/s-u/proj4/issues/6

adokter, 2020-05-08:
See adokter/bioRad#385 and https://docs.ropensci.org/rnoaa/ to read NEXRAD data into R directly