adokter/vol2bird #142: Full install, does not like odim files

ce88888, 2020-03-29:
Hi there

With much trouble I followed the install directions and eventually got everything to build/install. However, when I try to read a nexrad file, it tells me that it can't read odim.

running ./vol2bird --help shows

vol2bird version 0.5.0 (11-Nov-2019)
   usage: ./vol2bird <polar volume> [<ODIM hdf5 profile output> [<ODIM hdf5 volume output>]]
   usage: ./vol2bird -i <polar volume or scan> [-i <polar scan> [-i <polar scan>] ...] [-o <ODIM hdf5 profile output>] [-p <ODIM hdf5 volume output>]
   usage: ./vol2bird --help

   Supported radar data formats:
   * OPERA ODIM hdf5 input format, see <https://www.eumetnet.eu/wp-content/uploads/2019/01/ODIM_H5_v23.pdf> [enabled]
   * input formats compatible with RSL, see <http://trmm-fc.gsfc.nasa.gov/trmm_gv/software/rsl> [disabled]
   * Vaisala Sigmet IRIS format, see <ftp://ftp.sigmet.com/outgoing/manuals/IRIS_Programmers_Manual.pdf> [disabled]

... this is after I did include rsl and iris .. so something gets ignored in the build.

... this is what I've used
/usr/test/src/vol2bird# ./configure --with-iris prefix=${RADAR_ROOT_DIR}/opt/vol2bird --with-rave=${RADAR_ROOT_DIR}/opt/rave --with-rsl=${RADAR_ROOT_DIR}/opt/rsl --with-libtorch=${RADAR_ROOT_DIR}/opt/libtorch --with-gsl=${GSLROOT}
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for libtorch library path... found
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking confuse.h usability... yes
checking confuse.h presence... yes
checking for confuse.h... yes
checking for cfg_init in -lconfuse... yes
checking gsl_multimin.h usability... yes
checking gsl_multimin.h presence... yes
checking for gsl_multimin.h... yes
checking for cos in -lm... yes
checking for cblas_dgemm in -lgslcblas... yes
checking for gsl_blas_dgemm in -lgsl... yes
checking rsl.h usability... yes
checking rsl.h presence... yes
checking for rsl.h... yes
checking for wsr88d_read_sweep in -lrsl... yes
checking for RSL installation... yes
checking for RAVE installation... found
checking existance of RAVE build configuration... found
checking for RAVE cc compiler... x86_64-linux-gnu-gcc
checking for RAVE cc options... -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -pthread
checking for cpp options... -DNDEBUG -g -fwrapv -O2 -Wall  -pthread
checking for RAVE ld flags... 
checking for RAVE shared flag... -fPIC
checking for Python include flag... 
checking for numpy include flag... 
checking for proj include flag used by RAVE... 
checking for proj library flag used by RAVE... 
checking for expat include flag used by RAVE... 
checking for expat library flag used by RAVE... 
checking if expat is supressed... yes
checking for hlhdf include flag used by RAVE... -I/usr/test/opt/hlhdf/include
checking for hlhdf library flag used by RAVE... -L/usr/test/opt/hlhdf/lib
checking for hlhdf config file... /usr/test/opt/hlhdf/mkf/hldef.mk
checking for hlhdf ld shared... x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-5Z483E/python2.7-2.7.17=. -fstack-protector-strong -Wformat -Werror=format-security  
checking for g++ ld shared options...  -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-5Z483E/python2.7-2.7.17=. -fstack-protector-strong -Wformat -Werror=format-security
checking for hdf5 software used by hlhdf... -L/usr/lib/x86_64-linux-gnu/hdf5/serial
checking for zlib software used by hlhdf... 
checking for szlib software used by hlhdf... 
checking for RAVE module libraries... -lravetoolbox -lhlhdf -lproj -lravetnc
checking iris2odim.h usability... yes
checking iris2odim.h presence... yes
checking for iris2odim.h... yes
checking for populateObject in -liris2odim... yes
checking for Vaisala IRIS support... yes
checking rb52odim.h usability... no
checking rb52odim.h presence... no
checking for rb52odim.h... no
checking for populateObject in -lrb52odim... no
checking for SELEX Rainbow support... no
configure: creating ./config.status
config.status: creating def.mk
config.status: creating libmistnet/configure
configure: starting CMake configuration of libmistnet
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/test/src/vol2bird/libmistnet

... any ideas?
Thank you.

ce88888, 2020-03-30:
I seem to have overcome this by starting from scratch and, in some cases, separating the && commands in the install.md file .. strangely, now iris2odim does not work but I don't need that for now

closed 2020-03-30 by ce88888

adokter, 2020-03-30:
Hi @ce88888, good to hear you managed to install, let me know if you have specific suggestions to improve the installation instructions.