adokter/vol2bird #143: Strange error with newer Nexrad data - wsr88d_decode_ar2v: not found

ce88888, 2020-04-01:
Hi there

I installed vol2bird on a clean ubuntu box as per the instructions (except iris). For older Nexrad data it runs fine, but if I pick newer files (the ones no longer compressed as .gz on the AWS repository), it throws an error as per the below.

I tried to search the repository for wsr88d_decode to see if I can work out from the code what is going wrong, but had no luck.

Strangely, the same files work just fine on my MacBook with a docker installation?

Any help would be much appreciated.

Thank you!
C.

root@ip-172-31-88-112:/home/opt/vol2bird/bin# ./vol2bird /home/rstudio/data_raw/KFTG20180707_232513_V06 test.hp5
Warning: no user configuration file 'options.conf' found. Using default settings ...
sh: 1: wsr88d_decode_ar2v: not found
wsr88d_decode_ar2v not found, aborting ...
Segmentation fault (core dumped)
root@ip-172-31-88-112:/home/opt/vol2bird/bin#

adokter, 2020-04-01:
Hi @ce88888, your error suggests something went wrong with the installation of your rsl library (you can get a copy here https://github.com/adokter/rsl), which is used by vol2bird. This library contains an executable wsr88d_decode_ar2v, I presume it wasn't built or installed correctly. It could also be that the program isn't exposed in your $PATH. Let me know if this helps to solve your issue!

adokter, 2020-04-01:
You can check if wsr_88d_decode_ar2v is installed in your terminal by typing `wsr_88d_decode_ar2v`, it should give a usage message:
```
$ wsr88d_decode_ar2v
Usage: wsr88d_decode_ar2v [--stdout] <inputfile> [<outputfile>]
```

closed 2020-04-01 by adokter

adokter, 2020-04-01:
@ce88888 FYI vol2bird is also available as a Docker image at https://hub.docker.com/r/adokter/vol2bird/ and - in case you need the mistnet segmentation model - at https://hub.docker.com/r/adokter/vol2bird-mistnet (this image is much larger). Running vol2bird through Docker is almost as fast al compiling yourself, at least on Mac and Ubuntu, and it saves you the hassle of compiling yourself.

ce88888, 2020-04-01:
Thanks for the response. I see wsr88d_decode_ar2v in /home/opt/rsl/bin ... so it did get built, if I call it there I get the output you have above. I then added /home/opt/rsl/bin to the LD_LIBRARY_PATH ... but still get the same error (I already had /home/opt/rsl/lib because that's needed for the rsl library as per your instructions, but seems this programme ends up in /bin not /lib)

still, when I call v2b it gives me the same error (I confirmed that the above was properly added to the LD_Path). .... so I'm at a loss.

I had some issue getting docker to run properly on ubuntu and rstudio server ... seems a permissions issue in that rstudio doesn't seem to see docker running. As I'm not a unix / CLI user, I ultimately thought it easier to build everything ... albeit that this turned out harder than your instructions led me to be believe, but I managed. I would think that the overhead of docker being invoked for every single file adds up ... I got a few years worth of data to process.

Any ideas? Thank you.

ce88888, 2020-04-02:
Adriaan - one more question: When is it better to use mistnet vs. just running the algorithm on autoconf and let it strip out rain that way? Thank you.

ce88888, 2020-04-02:
@adokter ... bumping this so you get notified.

adokter, 2020-04-02:
Hi @ce88888, try adding  /home/opt/rsl/bin to PATH instead of LD_LIBRARY_PATH and see if that fixes your issue

adokter, 2020-04-02:
> When is it better to use mistnet vs. just running the algorithm on autoconf and let it strip out rain that way? 

I would say if you use historical non-dual polarization S-band radar, mistnet is your best option. In other cases (C-band, or recent dual-polarization data from 2013 onwards) you do not need it

adokter, 2020-04-02:
Regarding Docker: did you succeed in following these install instructions? http://adriaandokter.com/bioRad/#installing-docker. You need to give Docker access to your drives for it to work.

ce88888, 2020-04-02:
@adokter - Adding it to the PATH variable did the trick! Thank you. I'll also look at the docker route when I have the time. I'm only processing Nexrad data - likely from 2000, which I believe is S band, but dual polarisation? Thus I should be ok without Mistnet? Thank you again.

adokter, 2020-04-03:
Ok great! Pre-2013 you do need mistnet. We've done a lot of processing of profiles already for the US, I might be able to share some if helpful, just send me an email with details

ce88888, 2020-04-06:
@adokter - great, thanks - I sent you a message on the address on your website.