adokter/vol2bird #159: iPoints are difficult to decode

bart1, 2020-09-23:
I'm currently parsing the iPoints array. In case there are multiple scans with same elevation angle this is some what challenging. This is partially due to the ordering by height bin. A simple fix would be to add the `scan id` and `range gate id` as an additional column.

adokter, 2020-09-23:
Agree this is a hassle @bart1. The points array has been introduced at some point by Jurriaan Spaaks from eScience center as a simplification, originally everything was read from the sweeps directly.

Instead of working on the points array which I wouldn't mind deprecating at some point, you could aslo consider working on
 https://github.com/adokter/vol2bird/issues/129, i.e. read the flags and classifications of each gate as an additional `param` object when you load a polar volume, instead of having to deal with the points array.

adokter, 2020-09-23:
Instead of adding the gate classifications to the point array, preferably they would be added to a new scan parameter added to each sweep.

bart1, 2020-09-23:
I agree with you I was now working on a quick hack to get things working for initial exploration. I seem to have it figured out now. Long term #129 is indeed the solution, i just dont have the familiarity to solve this in vol2bird yet. (just in this exercise figured out the min and max ranges are indeed ranges (atleast in 0.3.20) and not distances as I kind of default assumed).