adokter/vol2bird #178: change vol2bird text output from fixed delimited to csv

adokter, 2021-03-26:


adokter, 2021-04-28:
vp-parser for R being developed here: https://github.com/enram/vp-parser

peterdesmet, 2021-04-29:
FYI, these are the metadata properties we discussed being relevant:

```
RADAR
/where/lon				5.176			[Double]	#Longitude
/where/lat				52.101			[Double]	#Latitude
/where/height				44.0			[Double]	#Height of the radar antenna [m] a.s.l.
/how/beamwidth				1.0			[Double]	#radar beam width [deg]
/how/wavelength				5.3			[Double]	#radar wavelength [cm]

FILE
/what/date 				"YYYYMMDD"		[String]	#Nominal Date (copy from volume file) -> link to data
/what/time				"HHmmss"		[String]	#Nominal Time (copy from volume file) -> link to data
/what/source				"WMO:xxxxx"		[String]	#Radar (copy from volume file)

SETTINGS
/how/clutterMap				"clutter.h5"		[String]	#Filename of cluttermap [string]
/how/dealiased				0			[Long]		#Whether dealiased radial velocities were used [0 no, 1 yes]
/how/enddate	 			"YYYYMMDD"		[String]	#Date (nominal date from last scan)
/how/endtime				"HHmmss"		[String]	#Time (nominal time from last scan)
/how/sd_vvp_thresh 			2.0			[Double]	#VVP radial velocity standard deviation [m/s]
/how/minrange				5.0			[Double]	#Minimum range [km] used for profile
/how/maxrange				25.0			[Double]	#Maximum range [km] used for profile
/how/minazim				0.0			[Double]	#Minimum azimuth used for profile
/how/maxazim				360.0			[Double]	#Maximum azimuth used for profile
/how/rcs_bird				10.0			[Double]	#Assumed bird radar cross-section [cm2]
/how/task				"vol2bird"		[String]	#Name of the program that generated this file
/how/startdate 				"YYYYMMDD"		[String]	#Date (nominal date from first scan)
/how/starttime				"HHmmss"		[String]	#Time (nominal time from first scan)
/how/task_args				"{see below}"		[String]	#Arguments of the program that generated this file
/how/task_version			0.2.2			[String]	#version of the program that generated this file
/how/comment				"Birds only"		[String]	#Indicator of how these profiles were generated
```