adokter/vol2bird #179: update to latest rave

adokter, 2021-06-01:
New rave release now available on github:
see https://github.com/baltrad

mkyleUD, 2021-11-11:
When trying to install vol2bird with the new version of RAVE, there is errors with a couple of the Rave functions called in libvol2bird.c. The two I have come across so far are with the `RaveIO_open` and the `RaveIO_load` functions. The errors returned are  `error: too few arguments to function ‘RaveIO_load’`. 

Looking at the `rave/librave/toolbox/rave_io.c` file it looks like Baltrad has add two more argument to those functions.
`RaveIO_t* RaveIO_open(const char* filename, int lazyLoading, const char* preloadQuantities)`
`int RaveIO_load(RaveIO_t* raveio, int lazyLoading, const char* preloadQuantities)`

I'm just sure what should be passed in place of those new variables when installing vol2bird.

adokter, 2021-11-11:
Thanks @mkyleUD, I noticed too recently and started a fix, which can install from this branch https://github.com/adokter/vol2bird/tree/rave_update - seems to be working ok but I need to do a few more checks and documentat updates before merging into the master branch

mkyleUD, 2021-11-18:
@adokter Thanks for directing me to the rave_update branch. I finally got some time to look at it again today and was able to get past the RAVE issue, but found another issue. In the linking vol2bird step this time. There seems to an issue with the "ABS", "MIN", & "MAX" functions.

------------------------------------
      linking vol2bird
------------------------------------
`
/usr/bin/c++    -o vol2bird vol2bird.o 
-L/lustre/it_css/sw/vol2bird_rad2/test-install/opt/rave/lib -L/lustre/it_css/sw/vol2bird_rad2/test-install/opt/hlhdf/lib  -L/opt/shared/hdf5/1.10.7/lib -L/opt/shared/proj/6.3.2/lib 
-L/opt/shared/proj/6.3.2/lib 
-L/lustre/it_css/sw/vol2bird_rad2/test-install/opt/rsl/lib 
-L/opt/shared/gsl/2.6/lib 
 
-L../libmistnet 
-lvol2bird -lravetoolbox -lhlhdf -lproj -lpthread -lm -lgsl -lrsl   -L../lib
../lib/libvol2bird.so: undefined reference to `ABS'
../lib/libvol2bird.so: undefined reference to `MIN'
../lib/libvol2bird.so: undefined reference to `MAX'
collect2: error: ld returned 1 exit status
make[1]: *** [vol2bird] Error 1
make[1]: Leaving directory `/lustre/it_css/sw/vol2bird_rad2/test-install/src/vol2bird/src'
make: *** [build] Error 2
`

mkyleUD, 2021-11-19:
@adokter After further testing it looks like the issue is proj. Vol2bird still requires proj 4.9, while the latest version of baltard/rave can use proj >=5. 

I'm trying to set up BioRad to use vol2bird outside a docker contain since I'm running on a HPC system. Since the BioRad install requires proj >4.9 it causes a conflict with vol2bird that requires proj 4.9. 

Is there any chance you can update vol2bird to use proj >=5?

adokter, 2021-11-19:
Thanks for looking into this - vol2bird uses proj only as a dependency of rave, so I presume this won't be a problem to update. I will look into this soon, not sure I can make it before the Thanksgiving Break though ...

mkyleUD, 2021-11-22:
Thanks for taking a look! I'm guessing the proj issue is related to the the change of name for the library from projects.h in versions <=4.9 to proj.h in versions <=5(or there about). I'm going to take a look again today, but if I find anything I will let you know. Thanks again!

adokter, 2021-11-22:
@mkyleUD did you also update hlhdf to the version that uses proj5? see https://github.com/baltrad/hlhdf

mkyleUD, 2021-11-22:
Yes I did. I try both yours and baltrad/hlhdf's branch. I was able install hlhdf and rave to use python3 and proj 6.3.2 but ran into the above mentioned errors.

mboscacci, 2021-12-22:
Just for information, after downloading latest hlhdf/rave version from Baltrad gthub repository, I did following changes on a RHEL8 Linux server with proj 6.3.2 library to successfully compile vol2bird:

`rhel8> **git diff lib/libvol2bird.c**
diff --git a/lib/libvol2bird.c b/lib/libvol2bird.c
index c0d5022..8aab658 100644
--- a/lib/libvol2bird.c
+++ b/lib/libvol2bird.c
@@ -3399,7 +3399,7 @@ radarDataFormat determineRadarFormat(char* filename){
     // try to load the file using Rave
     // unfortunately this loads the entire file into memory,
     // but no other file type check function available in Rave.
-    RaveIO_t* raveio = RaveIO_open(filename);
+    RaveIO_t* raveio = RaveIO_open(filename, 0, NULL);

     // check that a valid RaveIO_t pointer was returned
     if (raveio != (RaveIO_t*) NULL){
@@ -4451,7 +4451,7 @@ PolarVolume_t* vol2birdGetODIMVolume(char* filenames[], int nInputFiles) {

     for (int i=0; i<nInputFiles; i++){
         // read the file
-        RaveIO_t* raveio = RaveIO_open(filenames[i]);
+        RaveIO_t* raveio = RaveIO_open(filenames[i], 0, NULL);

         if(raveio == NULL){
             fprintf(stderr, "Warning: failed to read file %s in ODIM format, ignoring.\n", filenames[i]);
@@ -4570,7 +4570,7 @@ RaveIO_t* vol2birdIO_open(const char* filename)
     goto done;
   }

-  if (!RaveIO_load(result)) {
+  if (!RaveIO_load(result, 0, NULL)) {
     RAVE_WARNING0("Failed to load file");
     RAVE_OBJECT_RELEASE(result);
     goto done;
`
and

`rhel8> **git diff lib/libvol2bird.h**
diff --git a/lib/libvol2bird.h b/lib/libvol2bird.h
index ec1e920..13bd683 100644
--- a/lib/libvol2bird.h
+++ b/lib/libvol2bird.h
@@ -15,6 +15,16 @@
 #include <polarvolume.h>
 #include <vertical_profile.h>

+#ifndef ABS
+#  define ABS(x)        ((x<0) ? (-1*(x)) : x)
+#endif
+
+#ifndef MAX
+#  define MIN(a,b)      ((a<b) ? a : b)
+#  define MAX(a,b)      ((a>b) ? a : b)
+#endif
+
+
 // ****************************************************************************
 // Definition of standard parameters.
 // ****************************************************************************
`
(definition taken from projects.h / libproj-dev  4.9.2)